bluerobotics/qgroundcontrol #20: Joystick sticks swap functions after reconnect

jaxxzer, 2017-06-24:


jaxxzer, 2017-07-28:
Reports that on disconnect/reconnect of a joystick, the functions of the two sticks are switched afterward. Investigate.

dheideman, 2017-08-08:
"When the controller was turned off or lost bluetooth vertical switched to the left stick where forward/backward was and forward/backward switched to where vertical was. It did not happen the first time that we turned on the computer today, but it did happen anytime the joystick lost communication."

dheideman, 2017-08-08:
Could not reproduce.

Tested on:
- Early 2015 MacBook Air, OS X 10.10.5
- Samsung P08AFC, Windows 10 Home Version 1703

on versions:
- QGC v3.2.2 (Windows 10), QGC v3.2.3 (OS X)
- QGC Development HEAD:b2c3264

with:
- Logitech Dual Action
- XBox One controller, wireless
- XBox One controller, wired (using 360Controller 0.16.5 driver on MacBook Air)

The switching problem was not encountered as described.

However if QGC is unable to save/load the calibration parameters for a controller, usually because the controller is being used for the first time, QGC will load the default calibration, which on the tested controllers will yield the "switched" configuration while in TX mode 3.  This occurred when the 360Controller driver's settings were changed to emulate an XBox 360 controller while QGC was open (wired XBox One controller, OS X).  However upon proper calibration in XBox 360 mode the switching ceased and had never occurred immediately upon reconnecting the controller.

jaxxzer, 2017-09-07:
good.

closed 2017-09-07 by jaxxzer