bluerobotics/qgroundcontrol #27: Mac requires wifi connection to use ethernet interface

jaxxzer, 2017-07-24:
It is reported that a Mac must be connected to a wifi network in order for the telemetry connection to work on the ethernet interface. Investigate using current master and[ QGC 3.1.3](https://github.com/mavlink/qgroundcontrol/releases/tag/v3.1.3).

dheideman, 2017-07-25:
Could not reproduce error.

Tested with an Early 2015 11" MacBook Air running OSX 10.10.5, QGroundControl v3.1.3, connected over ethernet to the Pi through a USB to ethernet adapter.

Tests were performed with the following MacBook Air host computer Wi-Fi states:
 - Wi-Fi off, and
 - Wi-Fi on, but not connected to a network,
and with the following Raspberry Pi 3 companion computer Wi-Fi states:
 - Wi-Fi connected, and
 - WiFi disconnected.

The following sequences were tested:
 - Start QGC, power up Pi, connect Pi to host computer
 - Power up Pi, start QGC, connect Pi to host computer
 - Power up Pi, connect Pi to host computer, start QGC

Telemetry connection was successful in all tests.

jaxxzer, 2017-07-25:
Thanks Daniel.

closed 2017-07-25 by jaxxzer