bluerobotics/qgroundcontrol #74: Joystick defaults to TX Mode 2 if connected to QGC after ROV

dheideman, 2017-09-15:
1.) If joystick is connected to the computer after QGC has recognized/connected to the ROV, the joystick will switch to TX Mode 2.

2.) If the joystick is connected before QGC has recognized/connected to the ROV, it will use the saved mode (by default this is Mode 3).

Any changes to the TX Mode in case 1 will not be saved.  Case 2 will use the mode last saved while in case 2 (or the default Mode 3).  For instance:
- running through case 1, then restarting QGC in case 2 will see the TX Mode set to Mode 3
- running through case 1, changing TX Mode to 4, then restarting QGC in case 2 will see the TX Mode set to Mode 3, and
- running through case 2, changing TX Mode to 4, then restarting QGC in case 1, changing TX Mode to 3, then restarting QGC again in case 2 will see the TX Mode set to 4.

May be related to #20

jaxxzer, 2017-09-18:
https://github.com/bluerobotics/qgroundcontrol/pull/80

closed 2017-09-18 by jaxxzer

jaxxzer, 2017-09-26:
https://github.com/mavlink/qgroundcontrol/pull/5668