bluerobotics/qgroundcontrol #83: Disable auto-deleting videos by default

jaxxzer, 2017-09-20:


jaxxzer, 2017-09-20:
I think we need two things:

- [ ] disable this by default. It should be opt-in
- [ ] warn/ask the user before deleting anything (maybe lock out the recording button)

dheideman, 2017-09-22:
See [pr-disk-use-overhaul](https://github.com/bluerobotics/qgroundcontrol/tree/pr-disk-use-overhaul)

dheideman, 2017-09-22:
[#5693](https://github.com/mavlink/qgroundcontrol/pull/5693)

dheideman, 2017-09-25:
Reworked according to Don's suggestions in PR #5693.  Waiting on PR #5699.

See [pr-video-limit](https://github.com/bluerobotics/qgroundcontrol/tree/pr-video-limit)

jaxxzer, 2017-09-26:
Done for BR builds here https://github.com/bluerobotics/qgroundcontrol/pull/91

closed 2017-09-26 by jaxxzer

jaxxzer, 2017-12-18:
https://github.com/mavlink/qgroundcontrol/pull/5729