bluerobotics/qgroundcontrol #103: PS4 d pad buttons on indices 16 and 17

jaxxzer, 2017-10-05:
They cannot be used, we limit the number of buttons to 16.

jaxxzer, 2017-10-16:
[PS4 hand controller.pdf](https://github.com/bluerobotics/qgroundcontrol/files/1387853/PS4.hand.controller.pdf)

jaxxzer, 2019-03-19:
If this is a problem, the resolution will be to add it to SDL2 gamepad.txt that does the mappings for known controllers.

patrickelectric, 2019-03-20:
In my last test everything is available the only problem is that does not show all buttons:
16 is available but only shows up to 15;
[image]

The value in the PDF is also wrong.

jaxxzer, 2019-03-20:
We still need the confirm label then, and to check on windows/mac