bluerobotics/qgroundcontrol #122: Add dive information in video

patrickelectric, 2017-11-12:
It's possible to use `textoverlay` to add log information with the video while recording.
Similar to #32

patrickelectric, 2017-11-12:
@jaxxzer what you think about that ?

jaxxzer, 2017-11-13:
I've done this: https://github.com/bluerobotics/qgroundcontrol/tree/pr-video-text-overlay

It uses a ton of CPU to re-encode the video, so this implementation at least is not practical for most laptops.

patrickelectric, 2017-11-13:
76e9baa CPU problem solved, I'll send a PR ASAP :)

Williangalvani, 2019-03-25:
Note: We decided subtitles (advanced substation alpha subtitles) are the best approach to overlay data on video without re-encoding and still allowing to turn the overlay on/off.
The discussion is now taking place [here](https://github.com/mavlink/qgroundcontrol/issues/7286)

closed 2019-03-25 by Williangalvani