bluerobotics/qgroundcontrol #139: Outbound heartbeat interval becomes irregular when connected to multiple vehicles

jaxxzer, 2017-12-04:
This is an issue when using the waterlinked gps system.

patrickelectric, 2017-12-11:
To test:
    
> connect to waterlinked driver, by going to /waterlinked and restarting the driver with nothing entered in the text field, that will connect it to the simulation server (need internet connection) and then in mavproxy console on the pi 'watch heartbeat'

> The waterlinked driver connects to QGC as a second vehicle, to get the boat position on the map as well as ROV you can also just connect multiple sitl instances, but that is not sure.

patrickelectric, 2017-12-12:
`APM: MYGCS: 255, heartbeat lost` appears when using QGC and mavproxy in the desktop side.

patrickelectric, 2017-12-12:
This is not possible to reproduce using only mavproxy and SITLs.
1. Start master: `mavproxy.py --master=localhost:14550`
2. Run SITL 0: `sim_vehicle.py -I 0 -j4 -D -L RATBeach`
3. Run SITL 1: `sim_vehicle.py -I 1 -j4 -D -L RATBeach`
4. Send `watch heartbeat` and all cli.

No heartbeat lost reported.

patrickelectric, 2017-12-12:
No heartbeat lost running **underwater-gps**, **mavproxy --master** and **SITL** in the same time. @jaxxzer any idea ?

jaxxzer, 2017-12-19:
The issue is that QGC receives data on local port 14550 from two different ports on the same remote host. Because the links are stored in a map<host,port>, qgc just uses the last remote port it heard from each time it sends outbound comms on the link.

PR #146 (2017-12-20): Autoconnect to UDP port 14551
Fixes #139
closed 2018-01-18 by jaxxzer