bluerobotics/qgroundcontrol #158: Error when download parameters

patrickelectric, 2018-02-09:
When using appimage:
Frame→BlueRov2/Vectored→ Load Vehicle Default Parameters →Blue Robotics BlueROV2

FirmwareUpgradeLog: PX4 releases github download failed "Error during download. Error: 99"

@jaxxzer can you confirm this ?

jaxxzer, 2018-02-09:
Duplicate https://github.com/bluerobotics/qgroundcontrol/issues/13

Are you in a VM?

patrickelectric, 2018-02-09:
> Are you in a VM?

No, I'm running in my main OS.

jaxxzer, 2018-02-09:
Can you download firmware?

patrickelectric, 2018-02-09:
> Can you download firmware?

Yes

jaxxzer, 2018-02-09:
Well it's apparent that it works in some cases but not others.... We need to find the difference in those cases.

patrickelectric, 2018-02-09:
With: export QT_LOGGING_RULES="qt.network*.warning=true;qt.network*.debug=true" 
```
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv2_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: QSslSocket: cannot resolve SSLeay_version
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot call unresolved function CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function sk_num
No Internet Access
qt.network.ssl: QSslSocket: cannot call unresolved function SSLv23_client_method
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_library_init
```

It appears to be related with: antony-jr/AppImageUpdater/issues/7

jaxxzer, 2018-04-10:
@patrickelectric Can you please try `http` instead of `https` here: https://github.com/bluerobotics/qgroundcontrol/blob/3.2.4/src/AutoPilotPlugins/APM/APMAirframeComponentController.cc#L207

patrickelectric, 2018-04-10:
Good call, will try it.

patrickelectric, 2018-04-26:
Everything failed.
Will restart the investigations one more time.

jaxxzer, 2018-04-26:
how about putting on a different host
like s3

jaxxzer, 2018-04-30:
Any new developments @patrickelectric?

patrickelectric, 2018-04-30:
The windows problems appears to be solved by: https://github.com/mavlink/qgroundcontrol/pull/6354
The linux problem, as described [here](https://github.com/mavlink/qgroundcontrol/issues/6337) with Don.
The workaround on linux OS that do not use libopenssl-1.0 as default, is to install 1.0 and use `export LD_LIBRARY_PATH=openssl-1.0_path_in_your_system` to set the correct path to search for it.

jaxxzer, 2018-05-01:
can we cherry-pick the windows fix?

patrickelectric, 2018-05-01:
yep, they are pretty simple.
I'll create a branch in BR/qgc to test it soon.

patrickelectric, 2018-05-01:
New windows binary [here](https://ci.appveyor.com/api/buildjobs/fw8ql91npleccltn/artifacts/QGroundControl-installer.exe).
PR based over 3.2.4, https://github.com/bluerobotics/qgroundcontrol/pull/183. Commits are from QGC: https://github.com/mavlink/qgroundcontrol/pull/6354

jaxxzer, 2018-05-30:
As seen on R6:

[image]

patrickelectric, 2018-05-30:
Can we test with the original QGC to see if the problem is with our fork ?

jaxxzer, 2018-05-30:
Yes, but I'm not comfortable asking these particular folks to do that for us. Let's keep it in mind for when this pops up again.

patrickelectric, 2018-06-22:
We should cherry-pick this:
https://github.com/mavlink/qgroundcontrol/pull/6590

jaxxzer, 2019-03-19:
@Williangalvani @patrickelectric this one is ok to close?

patrickelectric, 2019-03-19:
Everything is working fine in upstream.

closed 2019-03-19 by patrickelectric