bluerobotics/qgroundcontrol #164: Application crash when opening popout

jaxxzer, 2018-03-07:


jaxxzer, 2018-03-07:
Only happens on certain computers, reports on Windows 10 and Mac.

dheideman, 2018-03-07:
Results of my tests:

| Computer | Operating System | Crashes? |
|--------------|------------------------|---------------|
| MacBook Air (11-inch, Early 2015) | OS X Yosemite 10.10.5 |  No issues |
| AMD Ryzen 7 1800X Eight-Core Processor x 16 | ubuntu 16.04 LTS | Crashes w/o error message |

patrickelectric, 2018-04-23:
Test version that uses [bluerobotics/patrickelectric/fix_video_popup ](https://github.com/bluerobotics/qgroundcontrol/compare/3.2.4...bluerobotics:patrickelectric/fix_video_popup?expand=1)
Download links:
- [macOS](https://s3.amazonaws.com/downloads.bluerobotics.com/QGC/builds/patrickelectric/fix_video_popup/QGroundControl.dmg)
- [Linux](https://s3.amazonaws.com/downloads.bluerobotics.com/QGC/builds/patrickelectric/fix_video_popup/QGroundControl.AppImage)

- [Windows](https://s3.amazonaws.com/downloads.bluerobotics.com/QGC/builds/patrickelectric/fix_video_popup/QGroundControl-installer.exe)

jaxxzer, 2018-04-23:
@dheideman please try this fix.

dheideman, 2018-04-23:
Testing, Part 2 (wip):

| Computer/Processor | Operating System | Crashes? |
|----------------------------|------------------------|---------------|
| MacBook Air (11-inch, Early 2015) | OS X Yosemite 10.10.5 |  TBD |
| AMD Ryzen 7 1800X Eight-Core Processor | ubuntu 16.04 LTS | TBD |
| AMD Ryzen 7 1800X Eight-Core Processor | Windows 10 Home 10.0.16299 | TBD |
| Intel Core i7-6700HQ CPU | Windows 10 Home version 1709 | No issues |

patrickelectric, 2018-04-27:
https://github.com/bluerobotics/qgroundcontrol/pull/176 Merged.