bluerobotics/qgroundcontrol #173: ~1 second + lag on video stream

vshie, 2018-04-22:
On latest QGC with stable build of ardusub, on both windows surface and macbook pro, significant lag and screen "tearing" of video present.
Telemetry stream does not suffer this lag.
Bandwidth verified as 60mbps both upload and download. 
The lag is present with or without video recording active. 
Sound stream does not seem to show lag, or as much of it. 

Lag present from initial connection, does not decrease or increase.

vshie, 2018-04-22:
Definitely not. I think it showed up around the last QGC update? Hard to pinpoint. Could be companion last companion update too. Are you able to replicate, or only have a linux topside?

jaxxzer, 2018-04-23:
I am not able to replicate this on windows or linux.

Can you try the previous releases to see if that is the problem? https://github.com/bluerobotics/qgroundcontrol/releases

In companion, you can move between tags on the /git page. Reboot companion after changing.

vshie, 2018-04-26:
On testing on Tuesday, had ROV temp sensor (in pixhawk?) reading 80+C with significant lag. Restarting QGC instance (latest, on Mac) eliminated issue. Connection with video pane open had been maintained for ~50 minutes, and lag developed from an initial typical level (100ms?) to 1 to 2 seconds. Repeating this test, enclsoure returned to 80C and opening it and blowing on camera reduced issue. 

Heat throttling of the camera processor is likely, but in such a condition would QGC potentially develop issues in handling of a stream? Especially one that is varying it's frame rate?

I'll focus on testing the previous releases in the same conditions today.

jaxxzer, 2018-09-20:
> Heat throttling of the camera processor is likely, but in such a condition would QGC potentially develop issues in handling of a stream? Especially one that is varying it's frame rate?

I have not noticed any issues while switching framerates, resolutions etc manually while the stream is running. This sounds like a hardware problem outside of qgc. Closing till confirmation.

closed 2018-09-20 by jaxxzer