bluerobotics/qgroundcontrol #213: Using Waterlinked and map window in QGC, without internet system is busy/latency

BoKoppel, 2018-10-08:
When using QGC map window for positioning with internet connection
through our ship network map works fine.
When we are offshore with no internet connection we get some network overload inside QGC computer.
If we zoom in the map, or move, QGC starts trying get a more detailed map and

 we get “lost connection” from QGC/Sub.
After some seconds we get connection back again, but  we have 1 to 2 secs of video delay.
Hiding the map window seems to fix it.
Possible solution: In QGC map setup, could there be a selection of “no map”?
Since we are working offshore we only need graphics of ship and of ROV with track, no actual map.

To reproduce:
Get external position to ROV
Use QGC without internet Connection and zoom in position on map.
/ Bo

Williangalvani, 2019-04-18:
[image]

Williangalvani, 2019-05-07:
So, it turns out that this issue is not as bad as it seems. This does not happen when offline.
This was very likely a issue with my own machine, as my internet often acts weird.
What I believe happens is that the [dns lookup](https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCTileCacheWorker.cpp#L1114) passes but the [connection itself](https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCTileCacheWorker.cpp#L1133) fails.

I [started to work on it](https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCTileCacheWorker.cpp#L1133), but now I'm not sure if such an specific case justifies that much new code. I still need to check what timeout is being used right now.

What do you think, @jaxxzer ?

jaxxzer, 2019-05-13:
@Williangalvani is unable to reproduce this after all

closed 2019-05-13 by jaxxzer